cu-ecen-aeld/final-project-PeterBraganza #4: UART application layer code

PeterBraganza, 2022-11-07:
Write an application layer for the GPS module. This application layer will be used with sample driver code and later will be used for the UART device driver.

DoD: 
1. Able to read GPS co-ordinates on the terminal.

suryakanteti, 2022-11-19:
New blockers from current sprint:
* [Verify UART and I2C patches](https://github.com/cu-ecen-aeld/final-project-PeterBraganza/issues/11)
* [UART Sample Driver](https://github.com/cu-ecen-aeld/final-project-PeterBraganza/issues/1)

suryakanteti, 2022-12-04:
Initially wrote logic to read multiple bytes of data (256) from the sensor. Turns out the UART driver we used, only gives 1 byte of data per read.

Modified the application logic to read 1 byte at a time until the end of one message ('\n').

suryakanteti, 2022-12-04:
The information needed from the GPS sensor is:
1. Latitude
2. Latitude direction
3. Longitude
4. Longitude direction
5. UTC time

There are multiple messages from the GPS sensor, but the one that gives us the needed information is **GPGLL**. The current code (https://github.com/cu-ecen-aeld/final-project-suryakanteti/tree/main/gps_application_code, commit ID e9c8638d68596fa8f30e46301c9463dadaa3a748) parses the GPS data for this message and stores it in a structure, prints it on the console.

The code is structured in a way that it can be used to extend to include the other message as well (GPGGA, GPGSA etc.)

suryakanteti, 2022-12-04:
Tested the built user-space application on the beaglebone board.

[image]

suryakanteti, 2022-12-04:
The UTC was earlier not as expected due to a parsing issue. Checked in the correction at commit ID afcbb9b5f82c5f2c003bf6e86d4ec37e1021380d.

closed 2022-12-05 by suryakanteti