cu-ecen-aeld/final-project-PeterBraganza #6: Socket code modification

PeterBraganza, 2022-11-07:


suryakanteti, 2022-11-14:
DoD:

Client Socket communication pair of applications where the message sent from one beaglebone board (client) end can be seen on another beaglebone board (server).

Blockers:

Issue https://github.com/cu-ecen-aeld/final-project-PeterBraganza/issues/9 can be a blocker while trying to test on beaglebone boards.

Alternatives approach:
Check the socket logic on two regular linux machines and test it on beaglebone boards once Issue https://github.com/cu-ecen-aeld/final-project-PeterBraganza/issues/9 is resolved.

Sona-Subair, 2022-12-05:
Deployed and verified socket communication. The code can be found under this [commit hash](https://github.com/cu-ecen-aeld/final-project-suryakanteti/commit/bc5df9fbc084c791687cf51d0051e7f6c58e3bd3). Screenshot of the captured is given below:
[image]

closed 2022-12-05 by Sona-Subair