cu-ecen-aeld/final-project-PeterBraganza #10: Enable I2C and UART

Sona-Subair, 2022-11-14:
DOD:
Enable I2C by adding patches and making changes to configuration

closed 2022-11-14 by Sona-Subair

Sona-Subair, 2022-11-14:
https://github.com/cu-ecen-aeld/final-project-Sona-Subair/tree/main/base_external/patches/linux

Sona-Subair, 2022-11-14:
Follow up:
[Verification](https://github.com/cu-ecen-aeld/final-project-PeterBraganza/issues/11)

dwalkes, 2022-11-17:
@Sona-Subair see discussion in class today - I don't think these will work on beaglebone.

Sona-Subair, 2022-11-17:
@dwalkes We are working on Buildroot 2019.05 version, we saw the verification results on the hardware support link.  We planned to do the verification this week  due to inconsistency in the buildroot image back then. Please see the link here https://github.com/cu-ecen-aeld/final-project-PeterBraganza/issues/11.