cu-ecen-aeld/final-project-PeterBraganza #12: I2C and UART sensor integration

PeterBraganza, 2022-11-21:
Application code which reads both I2C and UART sensor data and sends it over socket application

DoD:
 * Able to get both sensor data. 
 * Able to send it over LAN(socket connection) and see same data values on client side

Blockers:

suryakanteti, 2022-12-04:
Added function in the GPS module to populate a string with the required data at repo https://github.com/cu-ecen-aeld/final-project-suryakanteti, commit ID 5134368238371359144fe7ca716214e265e9d9ee.

suryakanteti, 2022-12-04:
Updated the application code to include this function and checked the string. Seems to be working.

suryakanteti, 2022-12-05:
Changed the function signatures in GPS application layer for ease of access with the socket code.

Repo: https://github.com/cu-ecen-aeld/final-project-suryakanteti
Commit ID: 4f93a1807bf1f7920172a1a7b8ad32fe989acb1f

suryakanteti, 2022-12-05:
Integrated on the server side and tested on the beaglebone. Seems to be sending the messages correctly. Since it was tested in the ESE lab, the GPS values were empty.

suryakanteti, 2022-12-05:
The current commit (74f30d6387a4e720adcaa67207602b7a40de6f92) is working. Need to send only the bytes in one message on server side to avoid filling the buffer frequently.

suryakanteti, 2022-12-05:
Integration done. Need to test it out on two different beaglebone boards.

suryakanteti, 2022-12-07:
Tested on two different beaglebone boards and used the home router for this. Able to see the sensor data sent from server to the client, where the client prints it on the console.

closed 2022-12-07 by suryakanteti